ruuvi/ruuvitag_fw #3: SDK patching

laurijamsa, 2015-12-04:
At the moment SDK requires some patching to compile example FW project correctly:

1) in SDK/examples/bsp/boards.h #elif defined(BOARD_CUSTOM) #include "custom_board.h" to #elif defined(BOARD_RUUVITAG_B1) #include "../../../ruuvitag_b1.h"

2) "Set the maximum number of characteristic client descriptors in the file device_manager_cnfg.h (located in \components\ble\device_manager\config):" #define DM_GATT_CCCD_COUNT 4

jrosberg, 2015-12-04:
why don't use custom_board.h ??
and #ifdef RUUVITAG_HW_REV_XXX inside that?
that way no patching (endless) is required.

laurijamsa, 2015-12-04:
Because no custom_board.h exists...

jrosberg, 2015-12-04:
how about creating one :smile:

custom_board.h
```C
#ifdef BOARD_RUUVITAG_B1
  #include "ruuvitag_b1.h"
#endif
```

and defining both `BOARD_CUSTOM`and `BOARD_RUUVITAG_B1`for build ?

the SDK includes are sooooo bad... (once again)

closed 2015-12-04 by laurijamsa