ruuvi/ruuvitag_fw #4: Enable GPIO high (current) drive for LED pins

laurijamsa, 2015-12-07:
To drive LEDs safely with more than 0.5mA current, high drive option should be enabled (for LED pins). High drive needs to be enabled on both, bootloader and FW.

http://infocenter.nordicsemi.com/topic/nrf52.v1.7/Chunk1498869550.html?cp=1_1_0_16_3_0#unique_1435213149

ojousima, 2016-10-11:
Moving this issue as task to trello, https://trello.com/c/tTjfWuW9. This issue will be closed on Github when the fix has been implemented.

ojousima, 2016-12-31:
Bootloader has high current drive enabled. I'd prefer to not enable high current drive on programs, as the difference in current consumption (2 mA vs 8 mA) is quite large and I've not noticed any adverse effects in driving LEDs with lower current. @laurijamsa comments?

ojousima, 2017-01-01:
Electrical specification says 1-4 mA (min / max) now.
[image]

Considering increased series resistance and that current specification is referred to voltage at pin (VSS + 0.4 V), I think the current drive can be left at "standard". Closing the issue.

closed 2017-01-01 by ojousima