ruuvi/ruuvitag_fw #28: Incorrect calculation of humidity?

samuelphy, 2017-03-20:
Since I don't know the details behind the humidity calculation for the weather_station I don't know which is correct but the comment don't match the code (line 173 in main.c).
`sensor_values.humidity = (uint8_t)(raw_h >> 11); 
 sensor_values.humidity <<= 2; //sensor_values.humidity = (uint8_t)((raw_h/1024) * 2);`

The code does: (raw_h / 2^11) * 2^2 = (raw_h / 2048) * 4
The comment says: (raw_h / 1024) * 2

I don't really see the point in throwing away accuracy at the senor side so my suggestion is to skip the two lines and just make one shift:
`sensor_values.humidity` = (uint8_t)(raw_h >> 9); `

samuelphy, 2017-03-23:
@ojousima pointed out that the reduced resolution data on URLs it to avoid spamming peoples phones (since they are notified for each URL change).
The comment and the code still does not match however and the comment could be a bit more informative about the reason behind the casting and shifting.

ojousima, 2017-04-08:
Fixed in #30 , thanks for reporting.

closed 2017-04-08 by ojousima