ruuvi/ruuvitag_fw #29: Uploading compiled applications to RuuviTag from Kickstarter with preinstalled bootloader

nikkekin, 2017-03-20:
Hey,

First of all, thanks for a great project. 

Being new to the project, I would just like to make sure I got it right and don't end up bricking the device (I don't have cables to to rescue, but I guess I will buy ones in case that happens :)). 

Anyway, I got the project compiled and I have a RuuviTag with the bootloader it was shipped with. I suppose I can now

*upload the sample applications I just compiled? Looks like I got v1.22 of the sdk, are applications compiled with this one compliant with the bootloader installed on my preinstalled Ruuvi?
*flash the bootloader, but are there instructions for this? 

I also noticed that make created packages under sdk11 and sdk12

ruuvitag_fw/builds/distribution_packages$ ls -l
total 0
drwxr-xr-x 1 vagrant vagrant 204 Mar 20 20:31 sdk11
drwxr-xr-x 1 vagrant vagrant 102 Mar 20 19:02 sdk12

Why is that?

I also noticed that at the same time the hex-folder contains still for example the weather_station.hex, which is actually in packages under sdk11:

ruuvitag_fw/builds/hex$ ls -l sdk12
total 280
-rw-r--r-- 1 vagrant vagrant 37548 Mar 20 19:02 ble_app_beacon.hex
-rw-r--r-- 1 vagrant vagrant 95336 Mar 20 19:02 ruuvitag_b3_debug_bootloader.hex
-rw-r--r-- 1 vagrant vagrant 95336 Mar 20 19:02 ruuvitag_b3_production_bootloader.hex
-rw-r--r-- 1 vagrant vagrant 46738 Mar 20 19:02 weather_station.hex

Also, perhaps I just missed this, but what exactly is pre-installed on the kickstarter RuuviTag? SoftDevice protocol stack + bootloader? Which versions?

Would be great if you had time to explain these so I can go forward.

-niklas

ojousima, 2017-03-20:
The SDK version is 12.2, softdevice version is 3.1.0.
SDK 11 builds are for historical reasons, I think they could be removed to avoid confusion. 

The bootloader shipped on Kickstarter devices is a debug version, which skips various version checks. This is intentional to allow people trying out their own software. 

Application shipped with kickstarter devices is SDK12 weather station.

You should update only application to be 100% sure to avoid bricking your device. 
To create a Device Firmware Update (DFU) package you should run:

`nrfutil pkg generate --debug-mode --application _build/nrf52832_xxaa.hex --key-file ~/git/ruuvitag_fw/keys/ruuvi_open_private.pem myapp.zip`

You can get _nrfutil_ from [Nordic](https://github.com/NordicSemiconductor/pc-nrfutil).

Then you can upload the package on your tag using the same process as described at [Ruuvi Setup Instructions](https://ruu.vi/setup ). 

Please ask Lauri for invitation to our slack channel (info@ruuvi.com) if you need to chat about building packages in detail

nikkekin, 2017-03-20:
Hey,

Sounds good, will give it a try. Actually already noticed the sdk11 versions didn't even upload :).

Thanks for a fast answer,
niklas

ojousima, 2017-03-20:
Happy to help, closing this issue.

closed 2017-03-20 by ojousima

nikkekin, 2017-03-20:
Just adding here, worked perfectly :)

Thanks,
niklas