ruuvi/ruuvitag_fw #33: LIS2DH12 Interrupt not working

pfafman, 2017-06-02:
Noticed that the driver for the LIS2DH12 says that the interrupt is not working.  Can someone elaborate or update what the status of this is?  Is it hardware or software?

Also the work around for power modes other than low and down does not work as implemented.  I think is is asking for an invalid clock tick for it fails at `app_timer_start`.  Seems the fastest the clock can go as set is around 400ms.

ojousima, 2017-06-03:
The issue is probably in SW, we simply never got around to implementing the interrupts as they're not needed in current SW. 

If you need a faster sample rate, please take a look at burst mode in https://github.com/ojousima/ruuvitag_fw/tree/feature-webble/drivers/LIS2DH12,

it collects the samples to accelerometer's FIFO and reads them out once a second (25 Hz). The update will be merged to master once the gatt profile is implemented and tested.

ojousima, 2017-12-04:
Please check merge-drivers pull request for working interrupts

closed 2017-12-04 by ojousima