ruuvi/ruuvitag_fw #37: Lincense unspecified ?

rzr, 2017-09-06:
I can read "Please note that these examples inherit a lot of code from various sources and pay careful attention to license and origin of each application."

It would help to have a top level LICENSE file that list term of use.

ojousima, 2017-09-06:
We'll compile the list of licenses. 
In the meanwhile, all of the applications and at least most of the drivers require presence (static linking) of Nordic Softdevice S132. The softdevice is closed source binary, so project is not GPL-compatible. as far as I understand GPL.

rzr, 2017-09-07:
Let me suggest to use debian's format:
https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/

closed 2018-05-03 by ojousima