ruuvi/ruuvitag_fw #41: DFU update problem

reyery, 2017-12-09:
I have downloaded the new eddystone_dfu_1.0.0.zip to update my device using nRF connect on my Android phone, however when the application has successfully transferred to the device, the red light is still on and I am unable to scan for it. I tried reflashing it multiple times via DFU however the problem still persists.

ojousima, 2017-12-09:
Thank you for the report. 
A few questions:
1) Is the model RuuviTag+ (with sensors) or one of the kickstarter units without sensors?
2) Have you used Espruino with your RuuviTag?

I can load the packet to my RuuviTag and it works as expected. However, the upload package does not erase configuration, such as unlock code. It's possible that your tag has some previous configuration, and Eddystone tries to parse it

ojousima, 2017-12-13:
@reyery Just a follow-up, could you answer the questions above?

ojousima, 2017-12-16:
Closing as stale. Please reopen the issue with additional information or contact us in slack if the problem remains.

closed 2017-12-16 by ojousima