ruuvi/ruuvitag_fw #67: Off by one error in ruuvitag_fw/drivers/lis2dh12/lis2dh12.c

jliphard, 2017-12-29:
The lis2dh12 has a size 32 FIFIO, but lis2dh12_get_fifo_sample_number returns 31 when the FIFO is full. Perhaps the value returned from LIS2DH12_FIFO_SRC_REG refers to the last _index_ or something like that? In any case, lis2dh12_read_samples(lis2dh12_sensor_buffer_t* buffer, size_t count) needs count = 32 to download the entire FIF0. Suggested fix (line 296 of lis2dh12.c):
```
*count = (ctrl[0] & LIS2DH12_FSS_MASK) + 1;
```

ojousima, 2017-12-29:
Thank you for the report. 
I'm looking at the lis2dh12 datasheet:
[image]

The datasheet would indicate that the register contains the number of unread samples, not the index.
Are you running the FIFO in STREAM mode, which will erase the oldest element on overflow? Is the behavior same on FIFO mode?

jliphard, 2018-01-09:
Hi @ojousima  - sorry for the delay. I changed to simple FIFO mode:

```
ctrl[0] = 0b01000000; //enable FIFO mode
write_register( FIFO_CTRL_REG_A, ctrl,  1);
```

and I then sampled at 10Hz and watched the FIFO fill up in real time without checking it. As with STREAM mode, `*count = (ctrl[0] & LIS2DH12_FSS_MASK)` maxes out at 31, one below the physical FIFO size of 32. I completely agree with your point above - in STREAM mode I could make an argument that lis2dh12_get_fifo_sample_number should never return more than 31, but in plain FIFO mode lis2dh12_get_fifo_sample_number should definitely return 32. Can you try to replicate on your end and confirm - this may be an issue with the LIS2DH12 firmware.

ojousima, 2018-01-09:
I tested the FIFO using master / test drivers examples, and I get 31 elements returned too. I'll look into drivers, maybe there is some misconfiguration somewhere.

ojousima, 2018-01-09:
Here is the result of a test run which tires to collect 35 samples:
```
 0> test_lis2dh12:INFO:FIFO had  0 samples, should be ~0 
 0> test_lis2dh12:INFO:FIFO had  31 samples, should be ~25 
 0> test_lis2dh12:INFO:FIFO had  0 samples, should be ~0 

 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1040

 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1040
 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1040
 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1040

 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1044
 0> test_lis2dh12:INFO:Got X:84, Y:56, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1040

 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1040
 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1036
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1036

 0> test_lis2dh12:INFO:Got X:76, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:60, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:60, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1040

 0> test_lis2dh12:INFO:Got X:84, Y:52, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:56, Z:1044
 0> test_lis2dh12:INFO:Got X:84, Y:56, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:84, Y:48, Z:1040

 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1040
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1044
 0> test_lis2dh12:INFO:Got X:80, Y:52, Z:1044
```

It seems that there are 32 unique samples, and samples 32, 33, 34 and 35 are the same (starting count at 1). Maybe the FIFO reports 31 samples in buffer, and sample number 32 is the last sample in current output register? 

Problem with adding one to the value returned by unread sample count would be that it would return "1" even when there is no new data to read. Personally I consider this as a "wontfix" right now and I think the behavior should be commented in the appropriate functions instead. What's your opinion?

jliphard, 2018-01-09:
I agree - "wontfix". A high read rates (read_freq >> sampling_rate) this is a non-issue because the FIFO will not be full, and adding one to the value returned by unread sample count will just corrupt the data trace by (re)reading the same values multiple times (<-confirmed on hardware).  At low read rates (read_freq << sampling_rate) the data will vanish into the ether anyway, whatever you do, and so if you really care about the integrity of the data trace you should have read_freq > sampling_rate. 

You could have lis2dh12_get_fifo_sample_number give an "UNDERREADING_WARN" condition when ctrl[0] & LIS2DH12_FSS_MASK starts to return values in the low 20s?  Then people would know that they are dropping data... But overall, best thing is simply to do nothing right now.

closed 2018-01-09 by jliphard