ruuvi/ruuvitag_fw #93: V1-backports RuuviFW 1.2.5: URL Beacon ID is NAN on ruu.vi site

ojousima, 2018-02-20:


ojousima, 2018-02-20:
This is caused by addition of "/" to ruu.vi URL, which bumps the ID char away. the "/" is required for some browsers.

Possible solutions: 
* Return to URL format 2
* Use ruuvi.com/, and replace .com/ with Eddystone shorthand to save one character.

@laurijamsa

Scrin, 2018-02-21:
Another possible solution is to remove the # from the url to save one character, though doing this will actually send the measurement to the server too as part of the request (the fragment identifier is not sent by the browser), and thus the page cannot be cached for changing measurements.

DG12, 2018-02-21:
@Scrin I previously proposed a complete solution which handles the  removal of the fragment identifier at weather-station-serverside

commit 6b4e3af1 (2018-02-25): fix #93
commit aeab9923 (2018-02-25): Merge pull request #100 from ojousima/v1-backports

Fix #93, #98
commit acd79191 (2018-03-01): Merge pull request #103 from ojousima/v1-backports

Fix #93, update version numbers
commit c6fb372d (2018-03-01): Fix #93
closed 2018-03-07 by ojousima