ruuvi/ruuvitag_fw #99: init error_code is ORed not ++. Also ...

DG12, 2018-02-20:
In main the statement that sets err_code=0 includes a comment that it is a counter. However the results from each initializer is OR with the last value of err_code.

During the main initialize section`drivers/init/init.c init_log` is called.
if` init_log` (i.e.` NRF_LOG_INIT`) errors it returns `INIT_ERR_UNKNOWN`
        (which is = 128 // Unknown error. aka 0x80)
main then `err_code |=` 

If nothing else fails init_blink_status is going to blink 128 times (really hard to count!)!

Also the do {...} while (init_status) will not exit since init_status is not changed.

This will disable the tag if it ever resets.

IMHO this entire set of logic needs to be reworked.

Scrin, 2018-02-20:
Since the tags have two leds (red and green) I would suggest to binary-code the status rather than blinking the led, for example the mentioned 0x80 is 1000 0000 in binary, which could be coded as one green followed by seven red blinks (or the other way around). 

Perhaps the error code indication could turn on both leds to indicate start of the sequence, then turn both leds off, and then start blinking the sequence, and finish it by turning both leds on again (or just a delay where both leds are off?)

DG12, 2018-02-20:
Please see my initial thoughts at
http://mybeacons.info/faultTolerance.html

ojousima, 2018-02-21:
There's *a lot* to clean up on master right now, I'll look into this issue once v1-backports release is official :)

DG12, 2018-08-03:
This issue as well as #102 are resolved with main.c revisions of 8/1/18

closed 2018-08-03 by ojousima