ruuvi/ruuvitag_fw #102: init_blink_status must always return

DG12, 2018-02-25:
main calls `drivers/init/init.c init_blink_status `after initializing each of the subsystems which - if an error has occurred - hangs with 

```
while(init_status);
    nrf_gpio_pin_set(LED_RED);
```
Although this is desirable for factory QA it is unacceptable for deployed tags which have been know to spontaneously reset.

Suggested change: On error stall for 10 seconds with red LED blinking  with 2:1 (on:off) duty cycle  to inform factory QA that a failure has occurred, 
then return to main.

ojousima, 2018-03-04:
I'll take this on my TODO, estimated to be completed in March 2018

DG12, 2018-05-03:
I'd like to take this on.

ojousima, 2018-05-10:
Thank you @DG12 for offering to help with this. Can you also take a look at #99 while you're at it?
Do you have a Ruuvi devkit already?

Please keep the error code as OR'ed, because the error codes will have a more precise meaning such as "device not found on bus" or "device self-test failed"

DG12, 2018-05-11:
OK

DG12, 2018-08-03:
This and #102 are resolved with update of main.c 8/1/18

closed 2018-08-03 by ojousima