ruuvi/ruuvitag_fw #106: Mode (raw/URL) must be saved so reset doesn't reset to default

DG12, 2018-03-03:
Tags have been known to spontaneously reset. This causes the tag to return to the mode (raw/URL) initial configuration.

Suggest saving mode in "UICR — User information configuration registers .. non-volatile memory"

CUSTOMER[0] 0x080 Reserved for customer 

NVMC must be enabled for writing in CONFIG.WEN ....a full 32-bit word ...write '0' to bits ..erased, that is, set to '1'.

See nRF52832 Product Specification V1.3, section 11 for details .
See api for proper function calls (this will be modified when I figure it out)

ojousima, 2018-03-04:
This feature has been on our TODO longer than I care to admit 😄 

I think UICR is not erasable during run time, [but only while flashing](https://devzone.nordicsemi.com/f/nordic-q-a/11781/writing-to-uicr-from-application-code). Please correct me if I'm wrong, maybe link to a devzone post or relevant code snippet?

Storing setup of the tag is definitely required by the time tags are configurable at runtime. I'm not 100% sure if the feature will be implemented in 2.X versions which are built on top of SDK12.3 or maybe in 3.X versions which will be built on top of SDK 14. 

I'll leave this one open as "help wanted", if some kind soul implements it and opens a pull request I'd be happy to merge the feature.

DG12, 2018-03-04:
Seems that UICR and other flash are independent.
 ERASEUICR is independent of ERASEPAGE  (of flash)
 however ERASEALL erases both.
Additional info:
See nRF52832 Product Specification V1.3, section  
11.3 Writing to user information configuration registers
14.1.28 CUSTOMER[0]  Address offset: 080 ,,,[31] 0FC

PS if the issue was worded "mode status lost during spontaneous reset" it could reconsidered BUG 😄

ojousima, 2019-01-04:
Done in e3785f004ffbf623d145944d3f4f62b88780a26e

closed 2019-01-04 by ojousima