ruuvi/ruuvitag_fw #110: Boot loader main must cripple WatchDogTimer

DG12, 2018-03-27:
WatchDogTimer reset:  As per nRF52832 Product Specification V1.3  "18 POWER section" "18.8 Reset behavior" "the WDT is retained at pin reset, Brownout reset and Power on reset". This means that if the watchdog time is started by the target application it will continue running after a reset. This includes during the MBR,  the boot loader process when the ruuvi tag is determining whether it should run the application or when the "B" button is depressed, enter the DFU(Device Firmware Update) mode. In addition WDT may expire during DFU. There is no stop of the WDT or feeding it  a large value which causes the DFU to sometimes be aborted when the WDT expires.

Although it would be best if the bootstrap  crippled the WDT, at least the ruuvi boot loader should.

Section 40.3 Watchdog reset states that ..." after a reset, or waking up from OFF mode, the watchdog
configuration registers will be available for configuration... 

WDT  INTENCLR should also be SET to "Disable interrupt for TIMEOUT event"

ojousima, 2018-04-05:
On a quick look it seems that bootloader in SDK15 feeds the watchdog. Therefore I'm marking this as wontfix here, and this issue will be resolved when we port to SDK15 (although this is far in future)

closed 2018-04-05 by ojousima

DG12, 2018-04-05:
Seems a simple enough fix affecting all tags distributed so fan and into the "far  future" not sure why you set to wont fix .