ruuvi/ruuvitag_fw #134: Ruuvi Firmware data encoding - data passed as argument is mutated

ojousima, 2018-08-09:
Make parameters to data encoding functions const

closed 2018-08-09 by ojousima

DG12, 2018-08-09:
Why?

ojousima, 2018-08-09:
It's a bug, fixed in #135

Scrin, 2018-08-10:
I find the current solution better, one struct for each distinct source of information, environmental contains all the environmental data from the BME280, acceleration contains all acceleration data from the LIS2DH12, and the remaining individual values comes from other distinct sources (interrupt counter, voltage, and configured tx power). It's bad practice to have too many arguments in one function, and using structs is a convenient way to avoid that.

For the "The real problem is that encodeToRawFormat5 has no business modifying a data item given to it.", that's what was fixed in #135 as it no longer changes the passed data, except obviously the (output) data buffer (=the encoded values are inserted)

DG12, 2018-08-10:
Warning platitudes follow:

There is a sensible limit on the number of arguments to be passed to a function but in this case using all simple arguments does not exceed it.

The other extreme is just make the variables global, then any function can reference(and modify) them without the bother of passing arguments (which I don't propose.)

I am among the (very) many that find the pass by value vs pass by reference indiscretion annoying at least.  

A better approach in this case is to provide all items in a single structure as is done in the (nearly next line) in the call to encodeToUrlDataFromat  including a sub-structure for various "classes" of items.
The inconsistency in the small area of code is unnecessarily confusing.

Perhaps much of the problem here lies with the multiple coders and various versions.

Providing data TO a function by reference is a poor choice, mixing them is a poorer choice. Adding the const qualifier (as promoted by many)  is a really incorrect approach (since the data is not constant)
Too bad there is no readonly qualifier. Oh wait, how about a macro
#define Read_Only const

And please avoid names like data and buffer.

Thanks for all the fish. %-)

commit 491c995d (2018-08-09): Merge pull request #135 from ojousima/master

Fix #134
commit e2353698 (2018-08-09): Fix #134