ruuvi/ruuvitag_fw #155: Improve battery handing

DG12, 2019-04-07:
There are two reasons for the RuuviTag to transmit the battery status:
1) Its value is interesting. After all, the RuuviTag is a collection of sensors and its main purpose it to sense things and report them.
2) To indicate whether the battery needs replacement.

There are several approaches to handling battery replacement.
a) Don't try to anticipate when the battery needs to be replaced. When the system fails, replace the battery. This approach is taken by many systems including things like the tire pressure monitors in automobiles and the lock key fobs.
b) Use heuristics and replace the battery after a number of months. The is the least disruptive  approach allowing the scheduling of the down time required to replace the battery. 
c) Send the current battery voltage to the client and expect the client to determine whether the battery needs to be replaced. This expects that the client has a better understanding of the system power usage of the battery than the ruuvi tag.
d) Have the ruuvi tag decide when the battery is nearing exhaustion. It has the best understanding of the power usage and can consider previous battery status.

Up through version 2.4.2 the approach has been to measure the voltage of the battery and transmit the value. This has several problems:
A) Transmitting the value takes precious bits ( 11 in the raw high resolution format 05).
B) Using the analog to digital converter is expensive from a power standpoint and is disruptive to the transmitter.
C) It depends on the client to analyze the voltage.

The following changes should be made:
I) Reduce the space required to transmit the battery status to 1 bit. To retain compatibility with current client code indicate battery OK by setting high order bit of byte 13 to 1. To indicate that the battery needs replacement the high order bit of byte 13 would be 0. This would make the 10 least significant bits of the format 05 battery voltage field available for future usage. As long as the other bits are unused, clients interpreting all 11 bits would be evaluate to values of 2.624 or 0.0. Future use of the other bits would cause current clients to interpret the values greater or equal than 2.624 ( i.e. 0xFFE0-0x8000 ) as OK and less as need replacement ( i.e  0x7FE0-0x0000)

II) Use the NRF52832's build in power supply supervisor to evaluate the battery voltage. Define the initial battery status as OK and set the POFCON register to 11. This sets the threshold to 2.4 volts. Once the threshold is triggered set the battery status as needs replacement.

See https://github.com/ruuvi/ruuvitag_fw/wiki/FAQ:-battery for additional discussion.

ojousima, 2019-04-08:
Tuning data formats requires co-operation from everyone who has ever implemented a library or plugin for RuuviTag, and therefore I'd prefer to avoid it.

The voltage limits are always dependent on use case, for example a tag in freezer might be at 2.3 V even though the battery is just fine. 

Having the power fail comparator trigger and storing the value at some point would be useful though, but the threshold and way to deliver data needs some additional thought.

Scrin, 2019-04-08:
>Tuning data formats requires co-operation from everyone who has ever implemented a library or plugin for RuuviTag, and therefore I'd prefer to avoid it.

Not only this, but it would cause a headache for all systems that store "original data". For example, the RuuviCollector does save the data format of all packets, altering the data formats after finalizing them will cause inconsistencies in systems like these. And not everyone updates their applications frequently. (assuming "tuning the dataformat" means modifying the existing one(s) rather than creating new, if creating a new format, then this doesn't apply as long as clients handle the data format identifier correctly)

As mentioned, the voltage can't be reliably used to determine the (remaining) battery life, even if you know the temperature. I've had tags run out of battery while providing *higher* voltage than before, at the same temperature. The voltage behavior also greatly varies between batteries.

>B) Using the analog to digital converter is expensive from a power standpoint and is disruptive to the transmitter.

This is not really an issue if it's done infrequently enough, the typical power consumption of a RuuviTag is already so low that the battery self-discharge is a much bigger issue when the measurement & transmission interval is long enough. Which should be the case anyway for use-cases where battery life matters.

This is especially true if comparing the power consumption of the radio itself; if extreme battery life is required, then a custom data format with the bare minimum amount of data should be used anyway, in which case a custom firmware for that particular use-case should be made using a custom data format.

Regarding the battery replacement itself; based on my experience across 8+ 24/7 active tags for 2+ years, there is absolutely no way to reliably determine when a battery needs replacement. The best guess is to estimate a very rough duration (in years, or months if using a power-hungry firmware with a small-ish battery) based on the firmware average power consumption, used battery and environment (temperature). Then just replace the battery as it runs out or starts having "odd behaviors" like spontaneous resets a few days to a few weeks prior to "fully running out".

DG12, 2019-04-09:
Thank you for your thoughts.

Regarding tuning existing formats. 
Remember that the code in the RuuviTag  is not like programs and apps for PCs and phones which announce new versions and implore (even threaten) the user to upgrade. I do not advocate revamping the existing format in such a way that existing clients cannot parse new or newly flashed tags. The "tuning" of the packet In a manner I have suggested would not break existing client code nor even be incompatible with existing data **format** (i.e it would still evaluate to a reasonable voltage value) even if existing tags were flashed with the new code. There is no reason that any historical data for a particular tag would become inconsistent.

Regarding the threshold at which point the RuuviTag defines that the battery needs replacement.
As Otso has mentioned, tags have been know to perform even with battery voltages as low as 2.3V.  Manufactures of Li/MnO2 batteries state that the minimum output is 2.0V. From the graphs that manufactures provide is impossible to interpolate how long it will take the battery to deteriorate from 2.0V until it stops supplying power at any given load and surely that depends on the load and perhaps temperature. Although RuuviTags may fail or become erratic at higher voltages(as Scrin has mentioned), it is reasonable to expect that at the 2.0V level the battery is **soon** to be exhausted.

Regarding the action to be taken.
I am not proposing that when the threshold is reached the tag shutdown or even stop transmitting. On the contrary, the tag should continue to transmit. It remains the responsibility of the client program to decide that an alert should be sent based on the battery condition and it is actually the user that needs to decide to when to replace the battery.

Although approaches to handling battery replacement described above as in **a** (wait for death) and **b** ( change based on time) are always an option, letting the user know that, based on the information available and the intelligence built into the RuuviTag, the battery will soon be exhausted seems to be a good additional option.

Perhaps this information is only useful for designing the next packet format.
Sincerely,
DG12

ojousima, 2019-09-03:
Closing this issue as stale. Maybe a future revision of firmware will use the brownout detector to flag the warning event in data format somehow.

closed 2019-09-03 by ojousima