ruuvi/ruuvitag_fw #156: 2.5.4: Invalid data on RAWv1

ojousima, 2019-09-03:


DG12, 2019-09-03:
One example is in bme280.c : compensate_H_int32 which will return 100% in error when calculation produces a value greater than 100! 
Should I open a separate issue for that?

ojousima, 2019-09-03:
I mean encoded wrong. Something broke when supporting the unavailable data for RAWv2, target is to fix it on 2019-09-04

DG12, 2019-09-03:
Regarding INVALID constants:
Rather than INVALID do you mean UNAVAILABLE ?

Although it does not make much difference,
you might consider using  more integer friendly values for example:
TEMPERATURE_UNAVAILABLE 126  // max is 85C , 126 is a small positive integer
HUMIDITY_UNAVAILABLE   126 // max is 100%, 126 is a small positive integer
PRESSURE_UNAVAILABLE  1101 // max is 1100
ACCELERATION_UNAVAILABLE 16001 // max per bme280 is 16000

Just a thought.

This code seems to imply that there may be ruuuviTags without either temperature/pressure/humidity or acceleration sensors or both.

commit 259f91b1 (2019-09-04): Fix #156
closed 2019-09-04 by ojousima