ruuvi/ruuvitag_fw #157: humidity from bme280 is incorrectly reported as 100% when calculated value is out of range

DG12, 2019-09-03:
The code in drivers/bme280.c is incorrect when the value derived from the sensor is outside of the range of 0%-100%.
If the calculated value is greater than 100% it incorrectly returns 100%.

The following code for compensate_H_int32 provides for 3 different options dependent on the initialized value :
1) Continue to incorrectly return 100%
2) Return an invalid value like 0% or 100.001%
3) Return most recent valid value

It requires a 4 byte static.
```
// set to an error value shifted 10 bits ( value returned includes 10 bit fraction) for example 100<<10+1.
//  or  100<<10 to return the legacy incorrect value of 100%
//  or 100<<12<<10  as a sample last valid value for first pass (incase first value is bad returns 100)
static uint32_t humidityCorrectness=100<<12<<10;
````
replace
````
    v_x1_u32r = (v_x1_u32r > 419430400 ? 419430400 : v_x1_u32r);    // 419430400 is (100 << 12) <<10
````
with
````
    If ( v_x1_u32r > 100<<12<<10 ) {  //  bad result
//                                1%<<12<<10 is lowest possible last calculated valid value
        if ( humidityCorrectness < 1<<12<<10)  return humidityCorrectness; // return a constant error value
         v_x1_u32r=humidityCorrectness;  //  use the last saved value
                                    }
    humidityCorrectness=v_x1_u32r;  // save valid  (or last valid ) calculation

  return  v_x1_u32r>>12;
````

The original code returns zero if the humidity is less than zero, not a problem since humidity can never be zero.

The most recent code at Bosch GitHub, since 08 Mar 2019 Version  3.3.6 is even worse as it incorrectly tests against 102.48% and sets result to 102.48!

I have not observed this problem with most recent tags I received.

ojousima, 2019-09-04:
As the RAWv1 and RAWv2 formats limit the reported humidity to 127.5 / 163.8, I'll limit the highest reported value to 120%. This allows user to apply offset on sensors which are showing too high values.

closed 2019-09-04 by ojousima

ojousima, 2019-10-31:
After some user feedback and discussion with Bosch the value is capped to 100% again

commit 7d008bc6 (2019-09-04): Fix #157