ruuvi/ruuvitag_fw #159: Bosch bme280 drivers now delays and checks for IM_UPDATE during init.  ruuvi_fw doesn't

DG12, 2019-09-15:
Recent revisions to Bosh bme280 driver seem to anticipate that the chip may not be ready as soon as expected.
For example: bme280_init calls soft_reset  which issues a SOFT_RESET, delays 2ms and then checks for IM_UPDATE. It loops here up to 5 times before returning. Init then reads the calibration data or delays 1 ms and retries up to 5 times !
From soft_reset function:

 ```
do                                                                                      
  {     
    /* As per data sheet - Table 1, startup time is 2 ms. */  
   dev->delay_ms(2);                       
   rslt = bme280_get_regs(BME280_STATUS_REG_ADDR, &status_reg, 1, dev);                
 } while ((rslt == BME280_OK) && (try_run--) && (status_reg & BME280_STATUS_IM_UPDATE)); 
        
```

The driver in ruuvi_fw does not seem to do this. 

Should it?

reference: [https://github.com/BoschSensortec/BME280_driver](https://github.com/BoschSensortec/BME280_driver)  around line 370 and 653

ojousima, 2019-10-16:
As the Ruuvi firmware has bootloader which delays the boot-up after initialization adding such a delay here isn't necessary. It definitely is a good thing for more general usage and 3.x firmware uses the official Bosch driver as a submodule and therefore contains this delay. 

As this codebase has been in use for over 3 years and we're not tracked a single issue down to this lack of delay (it's been checked on some units with faulty BMEs) I'm inclined to mark this as a wontfix.

DG12, 2019-10-17:
Since the current codebase doesn't seem to issue a bme280_soft_reset I guess it relies on the hard power to reset. 

Should it call bme280_soft_reset ?

ojousima, 2019-10-17:
> Should it call bme280_soft_reset ?
Not really as the configuration is static. Soft reset is useful when developing code as it allows the code to start up at a known state, but it's more an issue on devkits which are always powered and may have any configuration on them at the start of the program

ojousima, 2019-10-17:
We can close this issue without taking action as a No Fix Required.

closed 2019-10-17 by ojousima