ruuvi/ruuvitag_fw #160: Duplicate entry in bootloader production Makefile

DG12, 2019-09-22:
`$(SDK_ROOT)/components/drivers_nrf/rng/nrf_drv_rng.c \
`
is included twice in bootloader/ruuvitag_b_production/armgcc/Makefile

There seems to be an extraneous 
`$(SDK_ROOT)/components/drivers_nrf/rng/ \    `
Notice it's  a directory

ojousima, 2019-09-23:
Thanks for pointing this out. Can you open a pull request with the duplicates removed? I can do it too at some point if needed.

DG12, 2019-09-23:
I'm a little unclear on how to do that. I have a fork that is
"This branch is 5 commits ahead, 35 commits behind ruuvi:master."
Do I need to delete my fork, reform it, make the change and then issue the pull request?

ojousima, 2019-09-23:
You probably could sort the fork out with some rebases, but maybe it's easier if I remove the duplicates later